wuxx/icesugar #2: iCELink not found

yoyeung, 2020-07-05:
When i try to "make prog_flash" it cannot find the path... look like iceLink removed from the repro?

wuxx, 2020-07-05:
which example you are testing ?
you can also use 
$icesprog xxx.bin
 to program the bitstream.

yoyeung, 2020-07-05:
Thanks for the reply, but i got this issue in mac.
Can't find iCE FTDI USB device (vendor_id 0x0403, device_id 0x6010 or 0x6014).
So that should I use type C port or another one?
Thanks

wuxx, 2020-07-05:
icesprog = icesugar program
not iceprog

yoyeung, 2020-07-05:
[image]

wuxx, 2020-07-05:
it's icesprog, not iceprog, notice the 's'

yoyeung, 2020-07-05:
[image]
I should install some lib or?

yoyeung, 2020-07-05:
Oops...i got u...

wuxx, 2020-07-05:
the binary is at
https://github.com/wuxx/icesugar/tree/master/tools/
it depends libusb & hidapi

closed 2020-07-05 by yoyeung