wuxx/icesugar #3: any plan to support icestudio and apio?

jemisa, 2020-07-14:
It's' a visual editor for open FPGA boards, at https://github.com/FPGAwars/icestudio 
and apio https://apiodoc.readthedocs.io/en/stable/source/contribute/support_new_board.html ?

wuxx, 2020-07-15:
thanks for the advice, maybe support in future :)

jemisa, 2020-07-19:
Thanks.
I'm trying to do it myself.

I'm unclear on the difference between 
icesprog leds.bin
and 
cp leds.bin /media/mylogin/iCELink
Both do program

Can you explain what is the role of icesprog ?

wuxx, 2020-07-20:
there are two way to program, as you say, one is use icesprog, the other way is drag and drop, first one use usb hid to program. and the the latter use usb msc bulk to program.
drag and drop program speed is faster, because it use the usb bulk transfer.
and icesprog is multi-functional, it can probe/write/read the spi-flash.

jemisa, 2020-07-20:
Thank you for the answers.

Here is what I did to make it work for asio

STEP 1:
If you want to use icesprog,  put icesprog somehere in your path
add
 "icesprog": {                                                                                                                                             
     "command": "icesprog",                                                                                                                                  
     "args": ""                                                                                                                                              
   },
to 
apio/resources/programmers.json

if you prefer to use the usb method, you need to make a bash script and put in your path as icesugar-copy
#!/bin/bash
vars=( ` df | grep iCELink` )
ICELINK_DIR=${vars[5]}
cp "$1" "${ICELINK_DIR}"

but I found out that the directory is unmounted after one copy... so I prefer to use icesprog

STEP 2:
add
"icesugar-copy": {                                                                                                                                             
     "command": "icesugar-copy",                                                                                                                                  
     "args": ""                                                                                                                                              
   },
to 
apio/resources/programmers.json

then add
"icesugar": {                                                                                                                                             
     "name": "icesugar iCE40  by MuseLab",                                                                                                                   
     "fpga": "iCE40-UP5K-SG48",                                                                                                                              
     "programmer": {                                                                                                                                         
       "type": "icesprog"                                                                                                                                    
     },                                                                                                                                                      
     "usb": {                                                                                                                                                
       "vid": "1d50",                                                                                                                                        
       "pid": "602b"                                                                                                                                         
     }                                                                                                                                                       
   }   
to
apio/resources/boards.json
(if you prefer to use the icesugar-copy as programmer, change the "type" to be "icesugar-copy" (it should match the name in the programmers.json )

then install apio

to use:
cd src/basic/verilog/leds
cp the common/io.pcf in the current folder then
apio build --board icesugar
apio upload --board icesugar

aalku, 2020-10-11:
Thank you @jemisa 
I made it work on Windows but don't know how to correctly prepare it to make it distributable and I didn't do the copy on Linux part yet.

My changes to apio are here
https://github.com/aalku/apio

And I created this issue in order to get help with it.
https://github.com/FPGAwars/apio/issues/214

Any help is welcome. This is a nice and cheap board and it would be nice to get it officially supported.

aalku, 2020-10-13:
Could you please release icesprog source or make a windows (libusbK?) binary so we can add it as apio programmer in a new toolchain?

the copy method does not support specifying an offset as far as I know.

wuxx, 2020-10-13:
yes, already commit under tools/src/

aalku, 2020-10-13:
Thank you! I'll try to create a windows binary with mingw. Wish me luck. 😂❤

aalku, 2020-10-13:
I had to make a couple of tiny changes but I made it work on windows.
I'm quite confident the c change is compatible with Linux but I don't know about hidapi-libusb being called just hidapi arround.

[icesprog-mingw64.zip](https://github.com/wuxx/icesugar/files/5373138/icesprog-mingw64.zip)

aalku, 2020-10-24:
@jemisa I've got apio toolchain here:
https://github.com/aalku/toolchain-icesprog/releases/tag/v1.0.0

apio fork supporting it here:
https://github.com/aalku/apio

I'm working with FPGAwars to include it on icestudio but I don't know when will it be working by default on nightly release or when the next stable release will be available. There are some steps ahead.

Obijuan, 2020-11-15:
The support of the icesugar in icestudio is almost finished! Thanks to @aalku  Here you can see an screenshot of a simple test circuit:

[image]

It is available in the develop branch. It is supported in Windows and Linux. The next step is add support for mac

jemisa, 2020-11-23:
https://mobile.twitter.com/Obijuan_cube/status/1330902188797939713 announces its support.
Thank @Obijuan and @aalku

closed 2020-11-23 by jemisa