wuxx/icesugar #9: Source code for ICELink

mamoit, 2020-09-08:
Is the source code for `ICELink` open?
I can find it anywhere.
If not, is it a "not yet", or no plans on open sourcing it?
It is a great tool, but I'm having some weird intermittent problems after dragging in a new program.

wuxx, 2020-09-09:
Hi，the souce code of iCELink will open in the future，and can you describe your problem in detail？

closed 2020-12-14 by wuxx

mamoit, 2020-12-14:
I am sorry for the radio silence, it was an issue with Kubuntu 18.04, that when the board reset the old mount was kept behind, making it quite messy after a couple of iterations.
With Kubuntu 20.04 this does not happen anymore, so I think the issue can be overlooked.

Did you release the iCELink source? Or you decided not to?
(not judging here, I understand completely if you opt by not releasing it)

wuxx, 2020-12-15:
iCELink source code currently has no open source plan, the firmware may be opened in future. :-)

camrbuss, 2020-12-31:
Is there any plan to add features such as adjustable ICELink clock output on PA8? Is it fixed at 12MHz?

wuxx, 2020-12-31:
the clock output will be able to adjust on iCESugar-nano board (base on ice40lp1k-cm36), because the lp1k-cm36 pll not support well by open source toolchain, current iCESugar seems not very necessary to have this feature ? it's pll can works well.
[image]