wuxx/icesugar #13: 1.5 schematic: Hard to see PMOD1 and UART are connected via J5

rvalles, 2020-10-18:
After some head-scratching, I realized the reason pin 4 seemed to be shorted high was that it's connected to STM32's UART TX via J5. 

In the schematic, PMOD1 does clearly show how some of the pins are connected to the microusb port.

It would be better if it also shown that the relevant two pins are also connected to J5, in the same manner.

wuxx, 2020-12-15:
OK, Thanks for the good advice, later will update the schematic

WizardTim, 2021-09-07:
I agree, took me a few minutes to realize what was going on.

If there ends up being a V1.6 PCB it might be a good idea to highlight pins on the silkscreen that have multiple uses (USB, UART & J7/PMOD1 shared IO). Just using the invert option in Altium Designer should do the trick. Would suggest to users just starting out with it they should investigate PMOD1's IO further before using it.

Once I figured that out I loved using this dev board! The drag and drop programming is very convenient.