wuxx/icesugar #15: The board is not working that keeps showing a "Fail.TXT"

ms0488638, 2020-11-01:
Hi, I just receive the board and try to use the demonstration bin file.

But every time I drag the bin file to the ICELINK in file explorer.

The board doesn't read the bin file, but showing a "FAIL.txt" file, which indicates the following lines:

error: The transfer timed out.
type: transient, user

What am I suppose to do?

wuxx, 2020-11-01:
Hello, did you change the short bridge, and which system you are in use?

wuxx, 2020-11-01:
and the default program in flash is the led blink, which ls flash by drag-n-drop before ship, can you see it ?

ms0488638, 2020-11-01:
> and the default program in flash is the led blink, which ls flash by drag-n-drop before ship, can you see it ?

Yap, the LED keeps blinking.

I am currently using the Windows 10 Professional Version.
I still am downloading the virtual environment from your link.

ms0488638, 2020-11-01:
> Hello, did you change the short bridge, and which system you are in use?

Should I modify any jumpers of the board?

wuxx, 2020-11-01:
no, just keep the default

wuxx, 2020-11-01:
and can you check the device manager to see if some driver is missing

ms0488638, 2020-11-01:
I just my linux laptop, and it can successfully upload.

Maybe I will check on my WIN10 desktop later, and find out which part went wrong.

wuxx, 2020-11-01:
OK

ms0488638, 2020-11-01:
I wonder if there is any available driver for the "iceLink" but the default one automatically installed by Windows?

wuxx, 2020-11-01:
there are 3 USB devices in iCELink, USB HID, USB CDC, and USB MSC, for win10 should no need to install any other drivers.

ms0488638, 2020-11-01:
This is weird, I've tried on 4 different computers.
Only my Linux laptop can successfully upload the bin file, while my windows computer may lead to Fail.txt.

Windows computer may occasionally success, but lead to fail.txt for most time.

If the bin file is successfully uploaded on windows computer, a explorer can show the transmission bar while the D4 and D5 LED can be observed blink.

wuxx, 2020-11-01:
eh... the factory led blink program was drag-n-drop with win10 too. can it may be the permission settings of you win10 ? 
after program succ,  the iCELink disk will remount, you can see it displear and appear again some seconds later

wuxx, 2020-11-01:
and how you drag the bitstream ? I usually open the iCELink folder and drag the bitstream into the disk. or can use ctrl-c ctrl-v too.

ms0488638, 2020-11-01:
Hey, I think I found the reason.

After I change to another Type-C-to-USB-A, all issues are gone.

I guess something goes wrong with the attached one. 

Problem might be fixed, thanks in advance.

wuxx, 2020-11-02:
ok

closed 2020-11-02 by wuxx