wuxx/icesugar #20: icesprog is available in official MSYS2 repositories

umarcor, 2020-12-05:
During the last weeks, some PRs were proposed with regard to building and packaging icesprog (#16, #17, #18, #19). 
I'm glad to tell that icesprog is now available in official MSYS2 repositories:

- https://github.com/msys2/MINGW-packages/tree/master/mingw-w64-icesprog
- https://packages.msys2.org/package/mingw-w64-i686-icesprog
- https://packages.msys2.org/package/mingw-w64-x86_64-icesprog

Therefore, users on Windows can install icesprog through `pacman -S mingw-w64-x86_64-icesprog` (64 bits) or `pacman -S mingw-w64-i686-icesprog` (32 bits); and they are ready to go!

```sh
user@DESKTOP MINGW64 /d/icesugar/demo
$ icesprog -w leds.bin
flash offset: 0x00000000
write flash (104090 (0x1969a) Bytes)
write 0x00000000
write 0x00010000
done
```

/cc @juanmard

wuxx, 2020-12-05:
great  job, huge thanks !!

umarcor, 2020-12-12:
A couple of days ago, icesprog was added to a package group named 'eda'. That allows installing it together with other related tools. See https://packages.msys2.org/group/mingw-w64-x86_64-eda and https://github.com/hdl/MINGW-packages#usage.