wuxx/icesugar #21: Is there a 'full pack' available including the ICESugar and several PMODs?

umarcor, 2020-12-14:
I'm looking into buying an ICESugar along with multiple PMOD peripherals. However, I didn't find any pack including more than a single PMOD.

| ICESugar    | 29.54€ |   Pack  |  Alone  |
|---|---|---|---|
| PMOD-VGA    | 36.30€ |  +6.76€ |   9.12€ |
| PMOD-TFTLCD | 37.98€ |  +8.44€ |  10.81€ |
| PMOD-LED    | 33.76€ |  +4.22€ |  *not available?*    |
| PMOD-SWITCH | 33.76€ |  +4.22€ |   6.59€ |
| PMOD-AUDIO  | 35.45€ |  +5.91€ |   8.28€ |
|             |        | +29.55€ |  34.80€ + PMOD-LED |

- PMOD-LED seems to be available as a pack only, not alone.
- Buying PMOD parts alone is ~2.36€ more expensive that buying one of them together with the board. I don't know whether that's an offer as a pack, or due to shipping.

The only option to get a "full pack" seems to be:

(ICESugar + PMOD-LED) + PMOD-VGA + PMOD-TFTLCD + PMOD-SWITCH + PMOD-AUDIO = 68.56€

Is that correct? Will everything be shipped together?

wuxx, 2020-12-14:
here the alone address of PMOD-LED
https://www.aliexpress.com/item/1005001504777342.html?spm=2114.12010615.8148356.1.68dc552e6pxQFe
the shipping fee is caclate as $2.8.
and we have two LCD: PMOD-TFTLCD (2.8 inch with spi interface); PMOD-RGBLCD (4.3 inch with rgb parallel interface)
the full pack will on stock soon, please wait... about some hours ? :-)

umarcor, 2020-12-14:
That was so fast! Thanks for the link and for letting me know about PMOD-RGBLCD (I had overlooked that!). I updated the table above.

> the full pack will on stock soon, please wait... about some hours ? :-)

Or some days. No rush at all. Take it easy ;)

wuxx, 2020-12-18:
the full pack is already available on stock :-)

closed 2021-06-17 by wuxx