wuxx/icesugar #22: What is the purpose of J7?

umarcor, 2020-12-14:
Comparing the schematics of v1.4 and v1.5, I found the difference to be the addition of J7, which seems to duplicate 4 of the pins in PMOD1:

[image]

The [render](https://raw.githubusercontent.com/wuxx/icesugar/master/doc/iCESugar_render.jpg) and [photo](https://raw.githubusercontent.com/wuxx/icesugar/master/doc/iCESugar.jpg) of v1.0 and v1.3, respectively, show that J7 was not there before. Comparing to a [photo of v1.5](https://raw.githubusercontent.com/wuxx/icesugar/master/doc/iCESugar_1.jpg) it looks like the distance between PMOD2 and PMOD3 was increased.

My understanding is that J7 was added for PMOD-RGBLCD, which has 30 pins (2x15). Hence a female angled 2x3P connector such as the following needs to be soldered in J7 for using PMOD-RGBLCD:

[image]

I assume that the spacing between PMOD connectors in PMOD-VGA and PMOD-TFTLCD matches the "new" separation in the v1.5 board design, and not the previous. With regard to the [PMOD specification](https://github.com/wuxx/icesugar/blob/master/doc/pmod-interface-specification-1_2_0.pdf) it seems that v1.5 complies with the separation between PMODs (22.86mm or 0.90"). Therefore, "standard" peripherals with two PMODs can be connected to the ICESugar now. Say, [Pmod SSD](https://store.digilentinc.com/pmod-ssd-seven-segment-display/) or [Pmod motor driver](https://shop.trenz-electronic.de/en/TEP0002-02-Pmod-compatible-motor-driver-board-15A-0-30V?c=375). By the same token, the PMOD-VGA and PMOD-TFTLCD of Muse Lab Factory Store can be used with any other board providing two properly spaced PMODs. Is this correct?

wuxx, 2020-12-15:
yes, the distance of v1.5 is obey the specification, you can use other standard PMOD peripherals.
and the J7 is expose 4 more GPIO and 5V power supply, it's suit for PMOD-RGBLCD with rgb565 color.
a bad news:  you can not solder the 2x3P connector between two 2x6P pmod, it will get stuck.
so when somebody buy the pmod-rgblcd, we will send board with 2x15p connector like this.
[image]

umarcor, 2020-12-18:
Thanks! That makes sense...

Does the PMOD-RGBLCD module depend on some of the signals in the 2x15 connector being differential pairs? Or are all pins used as general purpose I/O?

It seems that the 5V in the RGBLCD are used for the backlight only. I.e., all signals are 3V3. Is that correct?

wuxx, 2020-12-18:
yes, the more 4 IO is perfect for rgb lcd with rgb565, check this demo for the detail
https://github.com/wuxx/icesugar/tree/master/src/basic/verilog/rgblcd
5V for the backlight (boost to about 18V), the IO is 3.3V.

closed 2020-12-18 by umarcor