wuxx/icesugar #24: Directly configuring iCE40 in slave mode.

EliteTK, 2021-01-09:
The "||" (prog iCE) jumper configuration allows the iCElink to configure the iCE40 in slave mode but there's no such option in icesprog. How would I go about having the iCElink directly configure the iCE40?

As a side note, the way the schematic names the busses is a bit confusing, I think ICE_MISO should really be called ICE_SO since it can operate in master and slave mode, same with ICE_MOSI (-> ICE_SI) this would make the schematic a lot clearer especially when in the default configuration (with the jumpers in "=" mode) ICE_MISO actually acts as ICE_MOSI and vice versa (the iCE40 is the bus master during the configuration).

wuxx, 2021-01-10:
when jump to "prog iCE", the spi-flash is bypass, the iCE40UP5k act as a spi slave device. this means we program the bitstream into internal SRAM of iCE40UP5k, and when system restart, the bitstream will disappear. currently this feature is not inplement yet. I think it's not a very important issue, some hardware design directly not allow to program to SRAM.

agree with your good advice, later will try to improve the schematic. :-)

wuxx, 2021-01-10:
if somebody really need this feature, may consider implement it in next version

EliteTK, 2021-01-11:
If there is already a command which can be sent over HID to the iCELink to do this then it would be enough to document it. I am already writing my own implementation of the icesprog tool (I wanted to clean it up and ended up re-writing it) and the way it is written would make it quite easy for me to add this feature.

JanDupal, 2021-04-05:
Same here – I've spent some time into digging how to enable SRAM configuration in icesprog before I had found this thread. It's not 100 % must-have, but having the option to do SRAM configuration would be great.