wuxx/icesugar #28: Question: Is there a USB PHY for the microUSB?

erlingrj, 2021-06-06:
Hi, and thanks for an awesome project. I have purchased an IceSugar FPGA board but have a question about the microUSB port. From inspecting the schematics it seems as if the D+ an D- pins are *directly* connected to IOB_18A and IOB_16A on the Lattice ICE40. I am not experienced in this domain, but I would have expected a USB PHY chip to be connected to the microUSB connector and a serial RX/TX interface of the USB PHY connected to the FPGA.

Am I missing something here?

wuxx, 2021-06-06:
Hello, the USB IO is just two normal GPIO, the USB 2.0 full speed  clk is only 6MHz, it's easy implement by fpga (even some MCU GPIOs, the software USB), you can check the demo code under advanced directory

erlingrj, 2021-06-07:
Thanks for your quick reply. I will check out the advanced directory. I found a guy that implements the USB stack at [TinyFPGA-Bootlaoder](https://github.com/tinyfpga/TinyFPGA-Bootloader), but that only implements the USB device endpoint, I need a host endpoint. But I will dig further, or I might just use a MCU with a USB Phy to act as a USB host. Anyways, thanks for your help. I will close the issue now.

closed 2021-06-07 by erlingrj