wuxx/icesugar #30: Drag&Drop fail with multi-image.

juanmard, 2021-06-17:
When you use a packet bitstream generated with **_icemulti_** and use a Drag&Drop _iCELink_ feature, always fail with a _time out_ error.

```
$ icemulti -v -p0 pwm.bin picorv32.bin leds.bin pll.bin -o icemulti_test.bin
Place image 0 at 0000a0 .. 01973a (`pwm.bin')
Place image 1 at 01973a .. 11b176 (`picorv32.bin')
Place image 2 at 11b176 .. 134810 (`leds.bin')
Place image 3 at 134810 .. 14deaa (`pll.bin')
```
Drag&Drop file _icemulti_test.bin_ and get this FAIL file:

[image]

wuxx, 2021-06-18:
what's the function of icemulti ? is it just conbine these bin file into a whole one image, like the cat command?
when drag-n-drop program, the on-board iCELink will check the file format, check the link here for detail
http://www.clifford.at/icestorm/format.html

juanmard, 2021-06-18:
Yes, **icemulti** is: _"IceMulti: A tool for packing multiple bitstream files into one iCE40 multiboot image file."_ 
http://www.clifford.at/icestorm (see icestorm tools)
or code https://github.com/YosysHQ/icestorm/blob/master/icemulti/icemulti.cc

Used to multiple _bitstream_: https://www.latticesemi.com/view_document?document_id=46502 (Chapter 10)
Not exactly like the cat command, because it include at the begining an «applet zone» with information about four _bitstreams_.

We would need to jump this applet to verify the format of the file and let it pass.

wuxx, 2021-06-18:
drag-n-drop need the file start with FF 00 00 FF, so it may only support program a single bitstream file, recommend to use the icesprog to program the multiboot image file

juanmard, 2021-06-18:
Ok, thank you.

closed 2021-09-07 by wuxx