wuxx/icesugar #32: Support for Lattice environment

lanczki, 2021-06-19:
It would be greate if the board could support the Lattice environment in a newer version.

What I'd like to propose is the following. 
The board should include a header where the iCE40 SPI pins and the CRESET_B pin are routed out.
[image]
[image]

Or just the CRESET_B,ICE_SS,ICE_SCK should need a header because the ICE_MOSI and ICE_MISO are already routed out to J3 
[image]

wuxx, 2021-07-07:
OK, will consider it later, thanks for you good advice~!

lanczki, 2021-07-13:
Thanks :) Keep up the good work.

closed 2021-07-13 by lanczki