wuxx/icesugar #36: could I use icesugar impelement VexRiscv

fatalfeel, 2021-09-21:
in https://github.com/wuxx/VexRiscv
VexRiscv small and productive (RV32I, 0.82 DMIPS/Mhz)  ->
    Artix 7     -> 232 Mhz 816 LUT 534 FF 
    Cyclone V   -> 155 Mhz 492 ALMs
    Cyclone IV  -> 155 Mhz 1,111 LUT 530 FF 
    iCE40       -> 63 Mhz 1596 LC

can icesugar impelement  VexRiscv?

hansfbaier, 2021-09-21:
@fatalfeel Yes you can and it has already been done: https://github.com/enjoy-digital/litex

hansfbaier, 2021-09-21:
@fatalfeel  And IceSugar is supported, see here: https://github.com/hansfbaier/litex-boards/blob/master/litex_boards/targets/muselab_icesugar.py

wuxx, 2021-09-22:
OK

fatalfeel, 2021-09-22:
thnaks my final goal is make risc-v cpu build on chisel3 scala with InteliJ IDE

fatalfeel, 2021-09-23:
> 
> 
> OK

plz add wechat friend: padmafeel 
thanks

fatalfeel, 2021-09-24:
after I built https://github.com/wuxx/VexRiscv/tree/d6b845ef8c79a002b43c6e6bd825e3213ac6a381
VexRiscv.v
Murax.v
how do I burn it to Icesugar, how to use it

my goal is burn a risc-v 
and 
a lite os 
and 
using riscv gcc build a simple app on it

fatalfeel, 2021-09-24:
got it ~~~will try later and report
root@homelinux:~/icesugar/icesugar/src/advanced/VexRiscv/scripts/Murax/iCESugar# make compile

closed 2021-09-24 by fatalfeel