wuxx/icesugar #45: Demo: uart_tx not working

piso77, 2022-01-03:
I tried both the prebuilt binary and to rebuild it myself - no output on the usb-c serial:

@cdone:1
@reset
@cdone:0
@prog [0x00000000]
@prog [0x00010000]
@start
@cdone:1

and if i connect the micro usb to my pc, no new serial devices appear (i'm using Linux, so no /dev/ttyUSB* show up) so i can't connect there.

How is one supposed to run this demo?

piso77, 2022-01-05:
Ok, after a bit of tinkering i realized what is wrong:

the microusb is not used at all, instead one is supposed to plug a PmodUSBUART (see Digilent website) into PMOD1, after that, and after opening a serial connection on ttyUSB0 @ 9600, i got the output.

But then i tried uart_echo, and i found this:

src/common/io.pcf:
...
set_io RX    4 # IOB_8A
set_io TX    6 # IOB_13B
...

but if you look at the schematics in schematic/iCESugar-v1.5.pdf, these two signals are on two different rows - oh, and even after fixing pin numbers, the uart_echo example still doesn't work.

LHLaurini, 2022-11-13:
You need to set the correct baud rate. Connect the board throught USB, then use `screen /dev/ttyACM0 115200`.

wil-low, 2023-01-26:
I got my UART working through USB-C with both jumpers on J5 in horizontal direction, and using /dev/ttyACM0 via minicom or Arduino IDE, with matching baud rate.

closed 2024-01-08 by wuxx