wuxx/icesugar #47: TFTLCD - SPI Register map

bartokon, 2022-03-20:
Hi, 
Where can I find datasheet with SPI register Map for pmod_tftlcd-v1.1?
Thanks.

wuxx, 2022-03-21:
hello
the lcd is based on ili9341, please check here
https://github.com/wuxx/icesugar/blob/master/doc/ILI9341_DS.pdf

bartokon, 2022-04-01:
@wuxx thanks for link!
Is there an example design for this LCD in the repository?

wuxx, 2022-04-02:
Hi, what do you mean the example design.
the example code can check here
https://github.com/wuxx/iceZ0mb1e/tree/master/firmware/main

bartokon, 2022-04-04:
Hey, Wuxx!

I have some questions regarding the LCD:
The "RS" is "D/CX" port? And the LCD is configured to 4-spi mode?
There is no other chip besides LCD on the PCB, is that normal? The LCD have inbuilt controllers?
Does touch functionality work somehow? If there is no controller? How?

Sorry for these questions, but the documentation and schematics are lacking some important information.

bartokon, 2022-04-05:
So, RS is D/CX, the LCD is configured in 4-spi mode. I still don't know about the touch functionality.

wuxx, 2022-04-06:
hello, RS is DC (data/cmd), LCD is in 4-spi mode. the LCD have inbuilt controllers.
currently the touch function is not supported on hardware

bartokon, 2022-04-06:
Hi, 
"LCD have inbuilt controllers ... **currently** the touch function is not supported on hardware". So that means LCD is touch-screen ready, but icesugar is not? 
I'm writing my own drivers for the LCD in Verilog, so if the LCD is touch-capable, can you provide me with a data sheet for touch driver?

wuxx, 2022-04-06:
Hello, touch is not currently supported on the hardware. Touch requires the circuit at the bottom of the board (based on xpt2046) and the support of the screen. All screens currently shipped do not have a touch screen installed.

wuxx, 2022-04-06:
when we mentioned touchscreen, the touchscreen include LCD display and a Touch panel on top of screen.

bartokon, 2022-04-06:
Okay! So last question that 3 little ic's are resistors/capacitors?

wuxx, 2022-04-06:
yes

MicroDuck, 2023-04-15:
hello good morning!
I wanted to know if you still don't have something about the TFT screen for the ICESugar_1.5

I am investigating to find out how it works and implement it in projects with Verilog, myself also to test the TFT modules that I can buy. First of all, thank you for your time and I hope to contribute something to you in the future.

bartokon, 2023-04-15:
> hello good morning!
> I wanted to know if you still don't have something about the TFT screen for the ICESugar_1.5
>
> I am investigating to find out how it works and implement it in projects with [...]

Take a look: https://www.hackster.io/bartosz-rycko/minized-with-pmod-tft-lcd-e0d682

wuxx, 2023-04-17:
hello, the screen is based on ili9341, there should be lot of document, just search the 'ili9341' in google

wuxx, 2023-04-17:
[ILI9341.pdf](https://github.com/wuxx/icesugar/files/11245933/ILI9341.pdf)
here is the spec, please check.

closed 2024-01-08 by wuxx