wuxx/icesugar #48: unable to run make prog

quantrpeter, 2022-12-10:
hi, my board is Lattice iCE40UP5k , when i plug it in, my ubuntu mount it as a drive and make prog not work, please help:

/home/peter/workspace/ice40_ultraplus_examples/leds>make prog
iceprog -S leds.bin
init..
Can't find iCE FTDI USB device (vendor_id 0x0403, device_id 0x6010 or 0x6014).
ABORT.
make: *** [Makefile:10: prog] Error 2

wuxx, 2022-12-10:
please try cmd icesprog leds.bin

closed 2024-01-08 by wuxx

brandcrz88, 2024-05-02:
Hi, I hope someone finds this useful.

I am using Ubuntu, for me the problem had to do with setting the rules and permissions properly.

- Stablish your PATH on your predetermined Shell:

export PATH="/path/to/fomu-toolchain-Linux/bin:$PATH"

- You can see that running "make prog" trows that err, this fixes that:

sudo /path/to/fomu-toolchain-Linux/bin/iceprog stopwatch.bin

- Now make that permanent. Check your device ID with:

lsusb

- If necessary, change your device rules, I had to change my idProduct from 6011 to 6010, and then add a GROUP="dialout".

- Go to:

sudo /etc/udev/rules.d/99-usb-serial.rules

- I changed:

SUBSYSTEM=="usb", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6011", MODE="0666"

- To:

SUBSYSTEM=="usb", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6010", MODE="0666", GROUP="dialout"

- Reload the rules with:

sudo udevadm control --reload-rules

sudo udevadm trigger

- Add user:

sudo usermod -aG dialout $USER

I hope that now you can run make prog directly from /stopwatch-dual