wuxx/icesugar #49: RGBLCD

djh82uk, 2022-12-24:
Hey, 

I have the larger PMOD LCD from the aliexpress store.  Looking at the src folder, the only example I can find is the rgblcd one.  However this does not work (pcf unconstrained errors).

I noticed the pin labels on the lcd board looks like it is using SPI (MOSI etc), while the PCF file in the example references Hsync, Vsync and pins for the different RGB colours, however the pin numbers do not all align with the  2 PMODS).  Am I right in thinking this example is not meant for the muselab TFT LCD PMOD board? 

If not, is there anywhere I can find examples or documentation to get started?

djh82uk, 2022-12-24:
AH, I did not realise there was an even larger LCD PMOD ([LINK](https://www.aliexpress.com/item/1005001505576467.html?spm=a2g0o.detail.1000013.2.75cd3284YWYJO3&gps-id=pcDetailBottomMoreThisSeller&scm=1007.13339.291025.0&scm_id=1007.13339.291025.0&scm-url=1007.13339.291025.0&pvid=191a2141-45d9-4f15-908e-2404f7e0c628&_t=gps-id:pcDetailBottomMoreThisSeller,scm-url:1007.13339.291025.0,pvid:191a2141-45d9-4f15-908e-2404f7e0c628,tpp_buckets:668%232846%238115%232000&pdp_ext_f=%7B%22sku_id%22%3A%2212000016385710159%22%2C%22sceneId%22%3A%223339%22%7D&pdp_npi=2%40dis%21GBP%216.77%216.77%21%21%21%21%21%40211b446116718402038474087e65a8%2112000016385710159%21rec)

I have the 2.3 inch one.  So I guess there just isn't any example code?  Any doc links would be appreciated.

Many Thanks

wuxx, 2024-01-08:
Hello, the rgblcd example in the repo supports 800x480 resolution.