wuxx/icesugar #58: iceSugar v1.5 Clock @ 6MHz

ButterBison, 2024-03-21:
I have just noticed that the ICE_CLK for my iceSugar v1.5 board is set at 6MHz. However the schematics indicate that the oscillator X1 is a 12MHz oscillator. 

The only thing I can see that can divide the cock is the icelink. 

I have tried with the iceprog tool as well to set the mco, but no luck.

```
$sudo ./icesprog -p
probe chip
board: [iCESugar]
flash: [w25q64] (8MB)
done

```

```
$ sudo ./icesprog -c 2
only iCESugar-Nano support mco select
^C
```

Can someone confirm this?

wuxx, 2024-03-22:
Hello, iCESugar's clock is driven by iCELink's PA8, and the default should be 12MHz. How do you measure the 6MHz clock currently, with a logic analyzer or an oscilloscope?

ButterBison, 2024-03-22:
Hi, 

Thanks for your reply. I just made a measurement with logic analyzer on the jumper J1.

[image]
[image]

[image]
 
The clock is correct. I will need to double check my code. Closing this issue for now.

closed 2024-03-22 by ButterBison