wuxx/icesugar #65: Unexpected RGB LED mapping on iCESugar v1.5 (Pin assignment mismatch?)

loadingStars, 2025-09-01:
# Unexpected RGB LED mapping on iCESugar v1.5 (Pin assignment mismatch?)

## Description
On the iCESugar v1.5 board, the silkscreen indicates the following RGB LED connections:

- **B** → Pin P39  
- **R** → Pin P40  
- **G** → Pin P41  

Based on this, I set up my Verilog like so:

```verilog
assign LED_G = 1'b1;
assign LED_R = 1'b1;
assign LED_B = 1'b0;
```

And my .pcf file contains:

```
set_io -nowarn LED_G 41
set_io -nowarn LED_R 40
set_io -nowarn LED_B 39
```

## Observed behavior
When driving ```LED_B = 1'b0```, the Green LED lights up.
When driving  ```LED_G = 1'b0```, the Blue LED lights up instead.

The mapping between pins and LED colors does not seem to match the silkscreen or the expected configuration.

## Expected behavior
Each LED color should correspond to its documented pin assignment without cross-activation.

## Questions

- Is this a known hardware quirk of the iCESugar v1.5 board?
- Could the silkscreen be mislabeled, or is there another explanation (e.g., common-anode/cathode wiring)?

### For reference:

```verilog

module switch (
    output wire LED_G,
    output wire LED_R,
    output wire LED_B
);

    assign LED_G = 1'b1;
    assign LED_R = 1'b1;
    assign LED_B = 1'b0; // Turn on Blue LED (but turns on Green)

endmodule

```

wuxx, 2025-09-03:
Hello, please try the following operation: remove all J6 shorting caps, bypass the FPGA, and then use a DuPont wire to short-circuit for LED lighting test. I tested on the board and it worked fine

loadingStars, 2025-09-03:
Thanks for the clarification and the J6 jumper-wire test method!
I confirmed it on my iCESugar v1.5 board: the silkscreen labeling for the RGB LED is incorrect. The actual working pin mapping is:
Pin 39 → Green
Pin 40 → Red
Pin 41 → Blue
The LED is active-low, so driving a pin 0 turns that color on.
So this means wiring on my board is wrong?

wuxx, 2025-09-03:
Hello, I'm sorry. I have checked the current products and found that the color of the old batch of boards is normal, but the color of the new batch is abnormal. It seems that there was a change in the batch sent to us by the LED manufacturer. We will adjust it back in the next version. I apologize for the inconvenience